SuLingGG/OpenWrt-Buildbot #6: 无法安装passwall

bmyjacks, 2021-04-14:
在最新的4-14版本中，nanopi-r4s-ext4镜像刷写后无法下载passwall。
[image]

SuLingGG, 2021-04-15:
可能是因为有关科学上网的某些关键的包编译失败了，可以后天再试试看。

SyllableSpec, 2021-04-16:
> 可能是因为有关科学上网的某些关键的包编译失败了，可以后天再试试看。

R4S 16号没更新嘛～

SuLingGG, 2021-04-18:
> R4S 16 号没更新嘛～

前几天编译遇到了一些问题，微调流程后已经恢复编译了，R4S 的固件已经恢复，可以下载最新固件或者 opkg update 然后安装你想要的软件包试试。

SuLingGG, 2021-04-18:
> 在最新的 4-14 版本中，nanopi-r4s-ext4 镜像刷写后无法下载 passwall。
> [image]

已修复。

closed 2021-04-19 by SuLingGG

bmyjacks, 2021-04-30:
[image]
again

Nlqqer, 2021-04-30:
@bmyjacks 你这个是哪个版本的？

bmyjacks, 2021-04-30:
> @bmyjacks 你这个是哪个版本的？

4-30

Nlqqer, 2021-04-30:
@bmyjacks 用个旧版吧，估计这个包某个依赖编译错了