SuLingGG/OpenWrt-Buildbot #9: 无法安装shadowsocksr

zyc530, 2021-04-17:
guring shadowsocks-libev-ss-redir.
Configuring shadowsocks-rust-sslocal.
Configuring shadowsocksr-libev-ssr-local.
Configuring shadowsocksr-libev-ssr-redir.
Configuring simple-obfs-client.
Configuring ca-bundle.
Configuring xray-core.
Configuring luci-app-ssr-plus.
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/shadowsocksr is different from the conffile in the new package. The new conffile will be placed at /etc/config/shadowsocksr-opkg.

SuLingGG, 2021-04-18:
我下载刷入了 openwrt.cc 的树莓派 4 固件，并没有出现你说的这个问题。
你是否进行过保留配置升级的操作？
或者你是否尝试了在已安装 SSR++ 的固件上升级 SSR++？
如果以上任一条件满足，请执行一次初始设置再试一试。

zyc530, 2021-04-18:
> 我下载刷入了 openwrt.cc 的树莓派 4 固件，并没有出现你说的这个问题。
> 你是否进行过保留配置升级的操作？
> 或者你是否尝试了在已安装 SSR++ 的固件上升级 SSR++？
> 如果以上任一条件满足，请执行一次初始设置再试一试。

zyc530, 2021-04-18:
是我这边的问题，更换机场后就正常了。

closed 2021-04-18 by zyc530