SuLingGG/OpenWrt-Buildbot #15: 最新版固件arch17。

xm3good, 2021-04-28:
兄弟们。，最新版本的arch17迷你固件，怎么刷不进去？
是不是固件的体积太大了？刷不进去？

Nlqqer, 2021-04-28:
@xm3good 善用搜索引擎

xm3good, 2021-04-28:
不是我不会刷固件，华硕arch我都刷了N遍了。，就是迷你版的这几天升级以后。我才刷不进去。，换成容量小的20多兆的就没问题。

SuLingGG, 2021-04-28:
我没有这个设备所以我不太清楚...
试试恢复一下出厂设置再试一次。
或者用命令行的方法升级试试 (命令行升级风险自担，方法请百度)

xm3good, 2021-04-28:
命令行升级风险太大了。，还是算了吧，我这是在opboot升级的，更新最新版本的总是超时。我怀疑应该是文件太大了，超出闪存的大小了。升级别的固件都正常。

SuLingGG, 2021-04-28:
> 命令行升级风险太大了。，还是算了吧，我这是在opboot升级的，更新最新版本的总是超时。我怀疑应该是文件太大了，超出闪存的大小了。升级别的固件都正常。

竞斗云也遇到过类似的问题，更新 opboot 到最新版可解。

xm3good, 2021-04-30:
G大，能不能把arch17固件私有安装包给去掉啊，我就是安装个纯净版固件儿，固件太大了，刷不进去。

SuLingGG, 2021-04-30:
> G大，能不能把arch17固件私有安装包给去掉啊，我就是安装个纯净版固件儿，固件太大了，刷不进去。

acrh17 的 rom 有 128M ，没问题。
你尝试 firstboot 一次，然后尝试把 opboot 更新到最新版试试看。
或者换用不同的浏览器打开 opboot。
我不想维护那么多 non shared packages，按目前的维护方式可能会在某个版本出现问题。

closed 2021-05-12 by SuLingGG