SuLingGG/OpenWrt-Buildbot #16: nanopi neo2 ipv6

longlongx, 2021-04-28:
固件似乎没有ipv6的相关模块，dhcpv6也没有
尝试opkg install ipv6-helper也拉取不到

SuLingGG, 2021-04-28:
chmod +x /bin/ipv6-helper
ipv6-helper install

longlongx, 2021-04-28:
IPV6 modules install successfully.

Configuring IPV6...

uci: Entry not found
uci: Entry not found
cp: can't stat '/lib/mwan3/mwan3.sh': No such file or directory
sed: /lib/mwan3/mwan3.sh: No such file or directory
IPV6 configure successfully.

longlongx, 2021-04-28:
谢谢！已经安装成功

longlongx, 2021-04-29:
另外想问一下
src/gz openwrt_core http://127.0.0.1/snapshots/targets/sunxi/cortexa53/packages
为啥默认的core源是本地的127.0.0.1，不是openwrt.cc啊

lxl6125, 2021-04-29:
> 另外想问一下
> src/gz openwrt_core http://127.0.0.1/snapshots/targets/sunxi/cortexa53/packages
> 这个是正常的吗
> 为啥默认的core源是本地的127.0.0.1，不是openwrt.cc啊

为最大程度保证兼容性，4 月 17 日之后的固件将会把所有 non-shared packages 软件包集成进固件内，之后的 non-shared packages 软件包将从本地软件源获取。
以树莓派 4 固件为例，固件体积将从原来的 40M 左右增大到 70M 左右。
(non-shared packages 软件包在固件中的存放路径: /www/snapshots)
如果你有条件，请尽快更新到最新版固件。
由于新版固件中所有的 non-shared packages 将从本地获取，预计 6 月 1 日后软件源将不再提供 non-shared packages 下载。

SuLingGG, 2021-04-29:
OpenWrt 用的 http 服务器叫 uhttpd，uhttpd 的根目录是 /www。

们用的 OpenWrt 控制面板叫做 LuCI，它相当于一个二进制文件，存放的位置是 /www/cgi-bin/luci。

由此可见，如果把某个文件放到 /www/archive/1.jpg ，那么在浏览器访问 http://192.168.1.1/archive/1.jpg 就能访问到这个文件。

所以我把所有 non-shared packages 软件包放到 /www/snapshots 下，那么访问 http://192.168.1.1/snapshots/* 就相当于访问到了这些文件。

而 OpenWrt 只需要访问自己就可以了，所以让它访问 http://127.0.0.1/snapshots/* 就可以了。

至于为什么要把 OpenWrt 的 non-shared packages 放到本地，是因为这些软件包相当于 OpenWrt 最核心的一些软件包，我之前采用的是在 OpenWrt.cc 下以日期做区别来处理的，这种处理方式有一个弊端，就是在固件跨天编译时可能会出现问题，比如在固件中写入的软件源日期是 24 号但是在源里的日期是 25 号。

此外，所谓的 non-shared packages 可以翻译为 “私有软件包” 意思是这些软件包是无法被共享，实际上 OpenWrt 的 non-shared packages 与内核配置有非常强的关联性。换句话说，同一波 non-shared packages 可能在几天或者十几天就没办法被别的固件用了，所以我在 OpenWrt.cc 做了与日期有关的处理。每天的固件只对应某个日期的软件包。

此外随着版本的不断增加，在 OpenWrt.cc 上必须存储每天的 non-shared packages。长此以往我必须维护大量文件，如果有某些文件缺失，就会导致大家在安装软件包的时候安装出错。

于是我把所有 non-shared packages 集成进了固件，这样既可以保持固件与 non-shared packages 的高度对应，不会出现不兼容的现象，同时也免去了维护大量版本的 non-shared packages 的问题，另外所有 non-shared packages 集成进固件只会让固件体积增加近 30M，这对于 ARM 设备来说是可以接受的。

longlongx, 2021-04-29:
$ opkg update
Downloading http://127.0.0.1/snapshots/targets/sunxi/cortexa53/packages/Packages.gz
*** Failed to download the package list from http://127.0.0.1/snapshots/targets/sunxi/cortexa53/packages/Packages.gz

但我今天似乎只有第一次update成功

SuLingGG, 2021-04-30:
> $ opkg update
> Downloading http://127.0.0.1/snapshots/targets/sunxi/cortexa53/packages/Packages.gz
> *** Failed to download the package list from http://127.0.0.1/snapshots/targets/sunxi/cortexa53/packages/Packages.gz
> 
> 但我今天似乎只有第一次update成功，但我从sftp去看目录，这个packages是存在的
> 我通过网页访问http://192.168.1.2/snapshots/targets/sunxi/cortexa53/packages/Packages.gz
> 也能够下载到Packages包

更新到最新版固件试试看？

closed 2021-05-12 by SuLingGG