SuLingGG/OpenWrt-Buildbot #24: r2s opkg openwrt_core 软件源问题

llllllllllx, 2021-05-15:
版本： immortalwrt-rockchip-armv8-friendlyarm_nanopi-r2s-ext4-sysupgrade.img.gz   21-05-14版
机型：R2s

执行`opkg update`报如下错误
[image]

会导致一些插件无法安装
[image]

尝试打开http://192.168.0.1/snapshots/targets/rockchip/armv8/packages
无法打开，所以判断问题在此处。
尝试将链接替换为http://OpenWrt.cc/snapshots/targets/rockchip/armv8/packages
重新执行opkg update 和 opkg install ..等操作后安装正常
不知道是不是r2s没有内置本地软件源导致？

SuLingGG, 2021-05-15:
你可以查看一下 /www/snapshots/targets/ 文件夹下是否有本地软件源。

SuLingGG, 2021-05-15:
另外你当前的这种更改本地软件源为 openwrt.cc 源的方法有效期只有两天，两天之后就会失效。具体表现为可以正常更新软件源，但是在安装 kmod 之类的 non-shared packages 时，会报内核版本不匹配的错误。

llllllllllx, 2021-05-16:
1, 本地是有内置源的，并且可以查找到`Packages.gz`文件
2. 使用`wget`进行下载，报错显示被拒绝了，那可能就是这里的原因导致最开始的报错信息。这种挂载目录是通过什么软件执行的呢？如何想要进一步排查原因的话该如何操作嘞？
3. 如果是使用`openwrt.cc`源有有效期的话，我直接替换成`file://`指向到本地源目录了,简单粗暴。  XD
[image]

Nlqqer, 2021-05-16:
@llllllllllx 改成接口的ip地址就好了

llllllllllx, 2021-05-16:
> @llllllllllx改成接口的ip地址就好了

目前是通过`file://`直接替换成指向到本地源目录了

我有一个猜测，是不是因为我是动态ip，然后被禁80了，所以被拒绝了。但是感觉又不应该呀，本地流量怎么会从外面走进来

SuLingGG, 2021-05-17:
感谢建议，本地软件源配置已于 ed054577706d56f04e25421e3029a4c9c729e6a7 修改为 file:// 协议。

closed 2021-05-17 by llllllllllx