SuLingGG/OpenWrt-Buildbot #26: 树莓派4B运行几个小时后，就会有一些后台页面无法访问了

tenkawataiyo, 2021-05-21:
每次重启后运行几个小时就会出现同样的情况，后台可以进，但是部分页面无法访问，而且能访问的页面修改配置也可能无法保存，错误提示如下

/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute arcombine dispatcher target for entry '/admin/network/network'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'admin_network/iface_overview'.
A runtime error occured: [string "/usr/lib/lua/luci/view/admin_network/iface_..."]:3: attempt to index a boolean value
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function

SuLingGG, 2021-05-21:
SD 卡可能寿命快到了。换张 SD 卡试试吧。

tenkawataiyo, 2021-05-24:
> SD卡可能寿命快到了。换张SD卡试试吧。

换了张全新的卡，问题依旧

lxl6125, 2021-05-24:
@tenkawataiyo 试过这个没
https://openwrt.cc/restore/

tenkawataiyo, 2021-05-27:
> @tenkawataiyo 试过这个没
> https://openwrt.cc/restore/
先刷还原固件问题依旧，这次跑了2天可能，今天早上进后台看，接口啥的页面又出现上面的错误信息了

tenkawataiyo, 2021-06-02:
找到问题原因了，passwall的UDP日志文件占满了/tmp分区，关掉就好了

closed 2021-06-02 by tenkawataiyo