SuLingGG/OpenWrt-Buildbot #30: 超大连接数

mingfengstudio, 2021-07-23:
OpenWrt-Mini，只安装了passwall和smartdns，且smartdns未开启，在开启一段时间后上级路由产生超大连接数（如图），且数字不断增加
OpenWrt-Rpi 试过也有相同情况

[image]

RyanShaw-3083, 2021-09-17:
Just relax! Try to enable the 'TCP MUX' options. (For your privacy, it is highly recommended that you blur the image before sending it out.)

mingfengstudio, 2021-09-17:
> Just relax! Try to enable the 'TCP MUX' options. (For your privacy, it is highly recommended that you blur the image before sending it out.)

Where can i enable TCP MUX?

RyanShaw-3083, 2021-09-17:
This is an upstream issue. When you configure server in `节点配置` an option at the bottom is the `mux`. If you want to control it in manual, edit your config files in directly.