SuLingGG/OpenWrt-Buildbot #33: Turbo ACC更改配置后保存BUG

tenkawataiyo, 2021-07-27:
<html>
<body>

型号 | Raspberry Pi 4 Model B Rev 1.1 (CpuMark : 34625.902536 Scores)
-- | --
架构 | ARMv8 Processor rev 3 (v8l) x 4 (1500Mhz, 46.1°C)
固件版本 | ImmortalWrt 18.06-SNAPSHOT r0-6131392 (2021-07-18) / LuCI openwrt-18.06-k5.4 branch (git-21.196.38136-1968c60)

</body>
</html>

系统日志：
> Wed Jul 28 05:58:02 2021 cron.err crond[1911]: crond (busybox 1.33.1) started, log level 5
Wed Jul 28 05:58:02 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jul 28 05:58:02 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jul 28 05:58:02 2021 daemon.info dnsmasq[2082]: Connected to system UBus
Wed Jul 28 05:58:02 2021 cron.err crond[2221]: crond (busybox 1.33.1) started, log level 5
Wed Jul 28 05:58:57 2021 daemon.err rmmod: module is not loaded
Wed Jul 28 05:58:58 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jul 28 05:58:58 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jul 28 05:58:58 2021 daemon.info dnsmasq[3602]: Connected to system UBus

修改Turbo ACC的设置后保存就会一直转圈，之后修改其他的系统设置可以保存，但不会生效，重启后可以恢复
修过防火墙设置添加端口转发也会出现类似的情况
[image]

dovela, 2021-08-01:
相似问题，我是卡firewall，修改无线配置时同样会转圈，但是修改本身是成功的。
树莓派3B+，版本ImmortalWrt 18.06-SNAPSHOT r0-83376b6 (2021-07-16) / LuCI openwrt-18.06-k5.4 branch (git-21.196.38136-1968c60)
[image]

EnnawYang, 2021-08-01:
遇事不决先重启，这个是系统本身的bug