SuLingGG/OpenWrt-Buildbot #34: 竞斗云启动无wifi

One971106, 2021-08-15:
附上log

One971106, 2021-08-15:
sys log 
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Sun Aug 15 13:42:50 2021 kern.notice kernel: [    0.000000] Linux version 5.4.132 (suling@ubuntu) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0-22968f0)) #0 SMP Fri Jul 23 05:32:02 2021
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] OF: fdt: Machine model: P&W R619AC
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Sun Aug 15 13:42:50 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 130560
Sun Aug 15 13:42:50 2021 kern.debug kernel: [    0.000000]   Normal zone: 1152 pages used for memmap
Sun Aug 15 13:42:50 2021 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sun Aug 15 13:42:50 2021 kern.debug kernel: [    0.000000]   Normal zone: 130560 pages, LIFO batch:31
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s30668 r8192 d22580 u61440
Sun Aug 15 13:42:50 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s30668 r8192 d22580 u61440 alloc=15*4096
Sun Aug 15 13:42:50 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129408
Sun Aug 15 13:42:50 2021 kern.notice kernel: [    0.000000] Kernel command line: ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1 rootfstype=squashfs
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] Memory: 506724K/522240K available (5902K kernel code, 192K rwdata, 1536K rodata, 1024K init, 242K bss, 15516K reserved, 0K cma-reserved, 0K highmem)
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Sun Aug 15 13:42:50 2021 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x2f0/0x4a0 with crng_init=0
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000007] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000020] Switching to timer-based delay loop, resolution 20ns
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000286] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000304] pid_max: default: 32768 minimum: 301
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000500] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.000516] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.001569] CPU: Testing write buffer coherency: ok
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.002568] Setting up static identity map for 0x80300000 - 0x80300060
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.002708] rcu: Hierarchical SRCU implementation.
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.003151] smp: Bringing up secondary CPUs ...
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.006236] smp: Brought up 1 node, 4 CPUs
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.006256] SMP: Total of 4 processors activated (384.00 BogoMIPS).
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.006263] CPU: All CPU(s) started in SVC mode.
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.011090] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.011253] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.011278] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.011490] pinctrl core: initialized pinctrl subsystem
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.012505] NET: Registered protocol family 16
Sun Aug 15 13:42:50 2021 kern.info kernel: [    0.012859] DMA: preallocated 256 KiB pool for atomic coherent allocations
[... cut ...]
Sun Aug 15 13:45:56 2021 daemon.err uhttpd[2026]: awk: /sys/class/ieee80211/phy*/device/hwmon/hwmon*/temp1_input: No such file or directory
Sun Aug 15 13:45:56 2021 daemon.err uhttpd[2026]: awk: /sys/class/ieee80211/phy*/device/hwmon/hwmon*/temp1_input: No such file or directory
Sun Aug 15 13:50:08 2021 daemon.err uhttpd[2026]: awk: /sys/class/ieee80211/phy*/device/hwmon/hwmon*/temp1_input: No such file or directory
Sun Aug 15 13:50:09 2021 daemon.err uhttpd[2026]: awk: /sys/class/ieee80211/phy*/device/hwmon/hwmon*/temp1_input: No such file or directory

One971106, 2021-08-15:
Kernel log
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.132 (suling@ubuntu) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0-22968f0)) #0 SMP Fri Jul 23 05:32:02 2021
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: P&W R619AC
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 130560
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] percpu: Embedded 15 pages/cpu s30668 r8192 d22580 u61440
[    0.000000] pcpu-alloc: s30668 r8192 d22580 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129408
[    0.000000] Kernel command line: ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1 rootfstype=squashfs
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 506724K/522240K available (5902K kernel code, 192K rwdata, 1536K rodata, 1024K init, 242K bss, 15516K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x2f0/0x4a0 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000007] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000020] Switching to timer-based delay loop, resolution 20ns
[    0.000286] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000304] pid_max: default: 32768 minimum: 301
[    0.000500] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000516] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001569] CPU: Testing write buffer coherency: ok
[    0.002568] Setting up static identity map for 0x80300000 - 0x80300060
[    0.002708] rcu: Hierarchical SRCU implementation.
[    0.003151] smp: Bringing up secondary CPUs ...
[    0.006236] smp: Brought up 1 node, 4 CPUs
[    0.006256] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.006263] CPU: All CPU(s) started in SVC mode.
[    0.011090] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.011253] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.011278] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.011490] pinctrl core: initialized pinctrl subsystem
[    0.012505] NET: Registered protocol family 16
[    0.012859] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.014081] cpuidle: using governor ladder
[    0.014182] cpuidle: using governor menu
[    0.014385] hw-breakpoint: Debug register access (0xee003e17) caused undefined instruction on CPU 0
[    0.014397] hw-breakpoint: CPU 0 failed to disable vector catch
[    0.014437] hw-breakpoint: Debug register access (0xee003e17) caused undefined instruction on CPU 1
[    0.014480] hw-breakpoint: Debug register access (0xee003e17) caused undefined instruction on CPU 2
[    0.014519] hw-breakpoint: Debug register access (0xee003e17) caused undefined instruction on CPU 3
[    0.015078] pstore: Registered ramoops as persistent store backend
[    0.015092] ramoops: using 0x80000@0x83f00000, ecc: 0
[    0.036125] cryptd: max_cpu_qlen set to 1000
[    0.039510] usbcore: registered new interface driver usbfs
[    0.039573] usbcore: registered new interface driver hub
[    0.039639] usbcore: registered new device driver usb
[    0.039687] pps_core: LinuxPPS API ver. 1 registered
[    0.039695] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.039721] PTP clock support registered
[    0.040521] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[    0.041696] clocksource: Switched to clocksource arch_sys_counter
[    0.042460] thermal_sys: Registered thermal governor 'step_wise'
[    0.042745] NET: Registered protocol family 2
[    0.042917] IP idents hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.043850] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.043893] TCP established hash table entries: 4096 (order: 2, 16384 bytes, linear)
[... 261 lines cut ...]
[   25.672932] Free swap  = 0kB
[   25.677955] Total swap = 0kB
[   25.681080] 130560 pages RAM
[   25.683963] 0 pages HighMem/MovableOnly
[   25.686809] 3623 pages reserved
[   25.694068] usbcore: registered new interface driver rt2500usb
[... cut ...]
[   31.362295] device eth0 entered promiscuous mode
[   31.362696] br-lan: port 1(eth0.1) entered blocking state
[   31.365989] br-lan: port 1(eth0.1) entered disabled state
[   31.371741] device eth0.1 entered promiscuous mode
[   31.379349] br-lan: port 1(eth0.1) entered blocking state
[   31.381369] br-lan: port 1(eth0.1) entered forwarding state
[   32.032459] pppoe-wan: renamed from ppp0
[   32.391785] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   32.392685] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

SuLingGG, 2021-09-16:
移除了冲突的驱动，最新的固件应该好了，可以再试一下。

closed 2021-09-16 by SuLingGG