SuLingGG/OpenWrt-Buildbot #38: X86_64 的 .gz文件有问题，解压提示文件损坏

JunyuMu, 2021-08-30:


SuLingGG, 2021-08-30:
在固件下载文件夹有提供 sha256sums 校验文件。如果 sha256sum 能对得上说明提供的文件没问题。

JunyuMu, 2021-08-31:
校验过，下载的文件没有问题。
主要是解压的时候。我是用的是Mac。解压会提示

[image]

使用终端解压:
```bash
$ gunzip immortalwrt-x86-64-generic-squashfs-combined.img.gz
gunzip: immortalwrt-x86-64-generic-squashfs-combined.img.gz: trailing garbage ignored
```
我搜索了一下，这个`trailing garbage ignored`可以忽略。
但我在把img 转成esxi的格式后，虚拟机无法正常加载。。。

我几个月前使用的此项目文件，没有出现过这些问题，解压都很正常。

可能是我哪里搞错了吧//

JunyuMu, 2021-09-09:
我这里测试了一下，发现是因为直通了网卡。
我的机器是HPE Gen10 Plus，网卡是Inter i350。

请问我这面需要怎么操作，是需要自己编译么？

SuLingGG, 2021-09-10:
> 我这里测试了一下，发现是因为直通了网卡。
> 我的机器是HPE Gen10 Plus，网卡是Inter i350。
> 
> 请问我这面需要怎么操作，是需要自己编译么？

不太清楚。我这边也没有 esxi 环境。
或者你可以用 qemu-img 转一下镜像格式再试试？

closed 2021-09-16 by SuLingGG